jasonacox/TM1637TinyDisplay #1: float decimal bug

davidboon78, 2020-08-09:
Hey so I love your library, and the animations are dope, but i'm trying to do a time display.  If I had a way to turn on the colon leds in the middle then I could just use the showString to display the time.  This would actually be preferable cause then i could make the colon blink on and off every second, but I couldn't figure out how to do that, it seems like you only show the colon by way of decimal points in the showFloat.  This works fine for double digit hours.  11.40 shows up fine as 11:40.  But 1.40 comes out as 140 with no semi colon.  Am I doing something wrong?

Thanks!

davidboon78, 2020-08-09:
my code looks like this if that helps:
display.showNumber(timeFloat,2,3,1);

closed 2020-08-09 by davidboon78

jasonacox, 2020-08-10:
Hi @davidboon78, thanks for the note.  I assume you figured out how to flash the colon.  The `showNumberDec()` function will help with this.  Below is an example I did to show a progress timing (mm:ss).  I have contemplated added a `showTime()` function/macro.  Let me know if you think that would be helpful.

```cpp
#include <TM1637TinyDisplay.h>

// Useful Time Macros
#define numberOfSeconds(_time_) (_time_ % 60UL)
#define numberOfMinutes(_time_) ((_time_ / 60UL) % 60UL)
#define numberOfHours(_time_) (( _time_% (3600UL * 24L)) / 3600UL)
#define elapsedDays(_time_) ( _time_ / (3600UL * 24L))

// Define Digital Pins
#define CLK 3
#define DIO 4

TM1637TinyDisplay display(CLK, DIO);

void setup() {
  display.setBrightness(0x0f);
}

void loop() {
  unsigned long timenow = (millis() / 1000) + (10UL * 60UL + 55UL); // for demo start at 10m:55s
  int days = elapsedDays(timenow);
  int hours = numberOfHours(timenow);
  int minutes = numberOfMinutes(timenow);
  int seconds = numberOfSeconds(timenow);

  if ((seconds % 2) == 0) {  // flash colon every other second
    display.showNumberDec(((minutes * 100) + seconds), 0b01000000, true); // turn on colon
  }
  else {
    display.showNumberDec(((minutes * 100) + seconds), 0b00000000, true); // turn off colon
  }
}
```

davidboon78, 2020-08-10:
I actually got it to work with just this (it alternates the colon every second)
		if (showColon) {
			timeFloat /= 100;
			display.showNumber(timeFloat, 2, 4, 0);
		}
		else {
			display.showNumber(timeFloat,0, 4, 0);
		}

But yeah that would have worked too!  Thanks!  Super useful library.  Thanks for the quick response.