jasonacox/TM1637TinyDisplay #2: Rotate display

nestorlevi, 2020-11-06:
Hello, can you create a function for 180 degree rotation? Congratulations on your work.

jasonacox, 2020-11-19:
Interesting.  That would require quite a bit of work.  I'll keep this in mind.  Pull requests welcome. :-)

ngmacha, 2022-05-07:
Hello, I would second this request too! Any update/progress? And what do you mean with "pull request" in this case - I have no experience on writing libraries... Thank you very much, anyway, for considering this request.

jasonacox, 2022-05-08:
Hi @ngmacha thanks for the feedback and upvoting @nestorlevi feature request.  It is possible, but I have some thoughts:
* The decimal or colon LED will be on the top and out of position.  Does this still make it usable?
* Mapping code will need to be added for segment LEDs as well as the sequencing (left to right for a 180 degree rotation) - TBD
* Would you expect to be able to dynamically flip the display in the program or would it be a compiler time choice (alway oriented one way or the other for your program)?  I ask because that could also help reduce the library size if there isn't a need to dynamically flip the display.

As I say, it can be done and will require work.  A "pull request" is how someone submits code to this project so it is just another way of me saying that I would welcome coding help if someone has time. :-) 

I do like this feature request, but I may not have time to get to it in a while.  Your interest does help.

ngmacha, 2022-05-08:
Hello @jasonacox , thank you for a quick reply - I've raised a same issue/request on https://github.com/bremme/arduino-tm1637/issues/56 and https://github.com/avishorp/TM1637/issues/92 but no reply yet, seems those two repo had not been working on for several years...

On your thought points:
- my display is https://www.seeedstudio.com/Grove-4-Digit-Display.html and doesn't have the decimal points, just a colon separating two digits on each side, totally symmetric if flipped 180 degrees.
- yes, but not sure how to do it in code?..
- dynamically flipping, depending on an Arduino pin value (details below) is preferably.

I'm trying to make a Kitchen Timer based on this one https://github.com/JAndrassy/KitchenTimerClock, and trying to cram everything into an old Blackberry charger. However the UK G-type power plug would allow it to be plugged, depending on the wall socket orientation, on 2 horizontally mirrored positions. My idea is to have an accelerometer to
 detect the orientation, and flip the display accordingly.

I like using your library, but tried to search everywhere if any of TM1637 libraries implemet this flipping feature, and found only here the mention of "inverted" parameter that allows flipping the display: https://www.esphome.io/components/display/tm1637.html. Don't know if that may help you...

I am not much of a coder, but ready to help to speedup the implementation of this feature, if you could tell me what need to do.

Thank you and kind regards - Mac Ha

jasonacox, 2022-05-08:
Thanks @ngmacha !  The use case helps.  This can be done in code and would be a nice feature to add.  Some notes and some requests:
* From the [TM1637 specifications](https://www.mcielectronics.cl/website_MCI/static/documents/Datasheet_TM1637.pdf), it doesn't look like the chip manufacturer, Titan Micro, has built in an easy "flip" command.  That means we will need to add code to support that.
* Need 1 - The digitToSegment map would need to be flipped (e.g. Seg A to Seg D) for each character. This needs to be done as a function instead of creating another memory map for `digitToSegment` and `asciiToSegment` - [see here](https://github.com/jasonacox/TM1637TinyDisplay/blob/35d1bdb24d6ad19e9c64257527423063cb22d40f/TM1637TinyDisplay.cpp#L33-L59).
* Need 2 - The ordering of the segment data sent to the chip needs to be flipped (eg. ABCD is sent DCBA) in the `TM1637TinyDisplay::setSegments` with the understanding that this command allows for position address starting point that would also need to be flipped. - [[see here](https://github.com/jasonacox/TM1637TinyDisplay/blob/35d1bdb24d6ad19e9c64257527423063cb22d40f/TM1637TinyDisplay.cpp#L191-L212)]
* Need 3 - Need to address decimal / colon setting as the way it is set is by adding `0b10000000` to the digit in front of the mark. This will need to be shifted to the adjacent digit for it to display correctly in the flip.

If you can help with any of the above, that would help us expedite the feature.  I need to keep the library small as it is meant for small (e.g. ATtiny85) microcontrollers as well as the larger ESP SoC units.

ngmacha, 2022-05-09:
Hi @jasonacox , for my particular use case I think having two different constructors - one that flips the display, the other same like current one, would be sufficient. For my timer changing orientation would require unplugging-plugging back the timer, which would reset the sketch and star it anew. If we select appropriate constructor within the setup() based on the value of the accelerometer,  nothing needs to be changed elsewhere in the sketch, am I right? Would this be easier to implement?

jasonacox, 2022-05-10:
@ngmacha and @nestorlevi  - I have added a flipDisplay() function that should do what you want.  

```cpp
// Includes
#include <Arduino.h>
#include <TM1637TinyDisplay.h>

// Initialize TM1637TinyDisplay - 4 Digit Display
TM1637TinyDisplay display(CLK, DIO);

void setup() {
  display.setBrightness(BRIGHT_7);
  display.flipDisplay(true);    // flip orientation of display
  display.showNumber(12.34);
  delay(1000);
}

void loop() {
  display.showString("HELLO");
  delay(500);
}
```

It is currently beta and only supports the 4-digit display module.  I'm running tests before releasing it as an Arduino library update.  However, I would love for you to test it.  You will need to pull the code down the repo code into your Arduino sketchbook library location (e.g. `~/Documents/Arduino/libraries` on a Mac):

```bash
cd ~/Documents/Arduino/libraries
git clone https://github.com/jasonacox/TM1637TinyDisplay.git
```

Let me know how it goes.

ngmacha, 2022-05-10:
Hi @jasonacox great job, it works!! I just had a chance to do a quick test, and all functions of my KitchenTimer sketch are working normally with a flipped display, just by adding a display.flipDisplay(true) to the setup()!

jasonacox, 2022-05-11:
Thank you @ngmacha !  That's great news.  I just completed the 6-digit display testing and plan to release this as v1.5.0.  Thanks for your help and patience.   

By the way, awesome idea for a project!  👍

ngmacha, 2022-05-11:
Hello @jasonacox , I've just got a response from @maxint-rd at https://github.com/maxint-rd/TM16xx/issues/20 - I've informed him about your implementation of the flip function, so he may like to learn and add into his library too. Hope you don't mind?..

Thank you too for a great support and excellent library. I think we may close this issue now?..

maxint-rd, 2022-05-11:
Hi All, interesting feature and good to know your use case!

Just my two cents: instead of an accelerometer, you might consider just a simple tilt-switch. These switches are glass tubes with a drop of mercury connecting to contacts if the switch is in one position, or disconnecting if it's in another. 

@Jason: I didn't know about your library yet, but once I find some spare time, I will have a look at it. Although my library isn't intended to be very small, I did use it on some smaller projects, including an ATtiny44 and a ATtiny85 based clock showing scrolling text on a [5x6 TM1637 led matrix](https://youtu.be/FfmRCfF5qQI).

jasonacox, 2022-05-12:
Hi @ngmacha, the change I made flips the 7-segment character maps (uint8_t) and inverses the data load sequence into the TM1637 registers - the changes can be seen [here](https://github.com/jasonacox/TM1637TinyDisplay/compare/v1.4.4...v1.5.0).  The explicit position setting was updated to factor in the inversion and the Dot/colon representation had to shift that to the adjacent segment to truly flip the display so that added more logic than just the bitwise shifts.   You might be able to make similar updates to https://github.com/maxint-rd/TM16xx in TM1637::send() or TM16xx::send().

Hi @maxint-rd!  I didn't know about your project either.  Very nice!   I started TM1637TinyDisplay because I wanted to be able to send alphanumeric strings (optional scrolling) and animation sequences to these simple 7-segment displays.  I had a project that needed that of course! :)  I also wanted to optimize it for the 8-pin ATtiny85 with additional modules (e.g. BME280), hence the "Tiny":  Low memory footprint, PROGMEM alphanumerics and animation sequences.

I tried to submit those changes to https://github.com/avishorp/TM1637 but they rightly declined my PR as the goal of their project is to use the 7-segment display for numeric output only. 

I have a web based animation tool to help build the data frames for animation:
https://jasonacox.github.io/TM1637TinyDisplay/examples/7-segment-animator.html

How would you describe the goal of your project?  I would like to add a link to your project from my README.

Cheers!

maxint-rd, 2022-05-13:
Hi @jasonacox , thanks for your reply and your kind remarks. I just now added your library to the links in my readme. I played with your online animator, which is very cool. Easy to use and quite handy. Well done!

The goal of my project is ease of use for the "entire" TM16xx chip family on the Arduino platform. 
The entire family is probably unachievable, but I think one has to set goals to get somewhere...

In fairness, it now supports only eight different chips and those are mostly the ones that I came across in various modules obtained from Ali. I started with just one module and the TM1638 library by RJ Batista (which now appears to be unmaintained). Then I got this cheap TM1637 module which was not supported yet. For reference I looked at avishorp's library and thought of a structure to support multiple chips. I ran across other modules, more chips and made some of [my own modules](https://github.com/maxint-rd/arduino-modules#display-modules). In the mean time I also wanted some easy to use features such as support for buttons and for the familiar print() function on both dot-matrix displays and 7/15 segment displays. As for size, I want to support tiny processors (ATtiny44A and up) but also larger MCUs (e.g. ESP32). That may become more challenging as things grow out of hand.

So far I've had years of fun with these chips and made a whole bunch of projects using these chips. Some I published on my YouTube channel: [youtube.com/maxint-rd](https://www.youtube.com/maxint-rd).

If I can find some spare time I want to take a closer look at your work, but that may take a while as I found that live is too short to do everything one may want to...

jasonacox, 2022-05-14:
Wow @maxint-rd that's amazing!  I love your YouTube videos and custom R&D boards... and robots!  Well done!!  

Thanks for the reference link.  I'm happy to add your project as well.  I completely agree on the life to short bit. So many toys, so little time. :-) 

Stay in touch!  I'll like and follow you too...

AIoT-Consulting, 2023-07-06:
Hi Jason,
Thanks for creating TM1637TinyDisplay.h, I’m using it for an application as per the pic bellow.
Because of the way the TM1637 board is constructed and due to very tight space limitations, I needed to position the LED upside down in my projects and use the flipDisplay() function.
I was wondering if any future updates you make might address the position of the decimal point  ?

Thanks again for this great library.

Richard
[image]

jasonacox, 2023-07-07:
> Thanks for creating TM1637TinyDisplay.h, I’m using it for an application as per the pic bellow.

Hi @AIoT-Consulting !  That's brilliant!  It looks GREAT!  I'm surprised at how many of my projects end up using `flipDisplay()`. Thanks for sharing this great photo!

> I was wondering if any future updates you make might address the position of the decimal point ?

Haha!  We could could it, `alterPhysicalReality()` and have it form a quantum tunnel to teleport the LED electronics to the opposite side. 😂  I enjoyed that, thanks, Richard.

FYI, for anyone following and confused, since we can't alter the physical electronics, flipping a decimal display vertically will always put the dot upside down.  However, all the display functions (e.g. `showNumber()`, `showNumberDec()`) that set the decimal will continue to work (just appear on the top) and you can move the dot to any horizontal position (or remove it entirely). 

https://github.com/jasonacox/TM1637TinyDisplay/blob/f13913e803ccfbe56f9ac924371738d19d66a597/examples/TM1637Test/TM1637Test.ino#L220-L224

AIoT-Consulting, 2023-07-07:
Awesome, if all else fails, we could also use a farcaster portal (Hyperion - Dan Simmons) to move the dot...
All kidding aside, I'll add some conditional code to move to dot using the example above...
Thanks Again !!!

jasonacox-sam, 2026-08-31:
This requested 180° display rotation. It shipped as `flipDisplay()`, which has since been used in lots of community projects — including upside-down mounted LED boards shared later in this very thread.

Closing since the feature has long been available.

— Sam 🔢

closed 2026-08-31 by jasonacox-sam