jasonacox/TM1637TinyDisplay #4: superfuous delay

nerdralph, 2020-11-17:
Only one delay per clock phase is needed, so the one on line 538 can be deleted.
https://github.com/jasonacox/TM1637TinyDisplay/blob/master/TM1637TinyDisplay.cpp#L538
Keep the delay on line 546.

Also the default of 100us seems excessive.  The datasheet v2.4 lists 500kHz for Fmax, which means 1us per clock phase.  I'd suggest a default bit delay value of around 5us, which would still leave a large safety margin.

jasonacox, 2020-11-19:
That makes sense, but when I drop the bitdelay to 5 or 10us, the Arduino Uno is unable to drive the display.   When I have more time, I'll investigate.

nerdralph, 2020-11-19:
Earlier today I hooked my scope up and found the rise time to 3V is nearly
10us with my module.  It has 10k pullups.  The datasheet example circuit
has 100pF caps, but the module must be using 1n.
I have been modifying my TM1638 library for the 1637.  I've avoided using
delays by polling the clk and dio lines when they get pulled high.

jasonacox, 2020-11-21:
I'm impressed!  Makes sense. I've seen a few other TM1637 libraries that use aggressive timing that caused stability issues over time for my project on some modules I use.  I may have over engineered the delays, but they are stable.  I'll revisit when I get time.  Thanks for all the feedback and PRs!

jasonacox-sam, 2026-08-31:
This suggested trimming the redundant per-clock-phase delay and lowering the default bit delay from 100us toward 5us (the datasheet allows 500kHz). Testing showed the aggressive timing didn't drive the display reliably on an Uno — scope measurements pointed at slow RC rise times on modules with weak pull-ups — so the conservative delays were kept deliberately for stability across modules.

Closing due to inactivity. If anyone runs these displays at tighter timings reliably across module variants, re-open with details — it'd be a nice speedup.

— Sam 🔢

closed 2026-08-31 by jasonacox-sam