jasonacox/TM1637TinyDisplay #9: How to show colon?

smartroad, 2021-04-07:
Hi! I have a display from eBay which has the 4 digits and the colon in the middle. I've been reading through but can't figure out how to get the colon to flash. I'm obviously missing something but I don't know what? Any suggestions?

jasonacox, 2021-04-08:
You can send the colon as a decimal value 0b01000000 using the showNumberDec() function.

Here is an example minute:second counter.  Would love to see your code.

```python
#include <TM1637TinyDisplay.h>

// Useful Time Macros
#define numberOfSeconds(_time_) (_time_ % 60UL)
#define numberOfMinutes(_time_) ((_time_ / 60UL) % 60UL)
#define numberOfHours(_time_) (( _time_% (3600UL * 24L)) / 3600UL)
#define elapsedDays(_time_) ( _time_ / (3600UL * 24L))

#define OFFSET (10UL * 60UL + 55UL) // start at 10m:55s

// Define Digital Pins
#define CLK 3
#define DIO 4

TM1637TinyDisplay display(CLK, DIO);

void setup() {
  display.setBrightness(0x0f);
}

void loop() {
  unsigned long timenow = (millis() / 1000) + OFFSET;
  int days = elapsedDays(timenow);
  int hours = numberOfHours(timenow);
  int minutes = numberOfMinutes(timenow);
  int seconds = numberOfSeconds(timenow);

  if ((seconds % 2) == 0) {  // flash colon every other second
    display.showNumberDec(((minutes * 100) + seconds), 0b01000000, true); // turn on colon
  }
  else {
    display.showNumberDec(((minutes * 100) + seconds), 0b00000000, true); // turn off colon
  }
}
```

smartroad, 2021-04-08:
Great, thanks :)

Here is my code:
```
    if (hour24) display.showNumberDec(hour(), second() % 2 * 0b01000000, true, 2, 0);
    else display.showNumberDec(hourFormat12(), second() % 2 * 0b01000000, false, 2, 0);
    display.showNumber(minute(), true, 2, 2);
```

jasonacox-sam, 2026-08-31:
This asked how to show/flash the clock colon. The answer: send it as dots data — value `0b01000000` — via `showNumberDec()`, toggled by `second() % 2` for the blink. The reporter got a working 24h/12h clock with exactly that.

Closing since the question was answered and confirmed.

— Sam 🔢

closed 2026-08-31 by jasonacox-sam