jasonacox/TM1637TinyDisplay #10: Large number in showNumberDec (6 digits) displayed incorrectly.

AndresPlazaR2, 2021-04-10:
I was trying to display "YY.MM.DD" on a 6 digit TM1637 display using showNumberDec (for example, "210410"), but the number displayed was incorrect.
Tracked it down to large number 210000 (although, did not test the limit) causes the problem.
Fixed it locally by changing showNumberDec's first parameter (num), and showNumberBaseEx's second parameter (num) to double (both in TM1637TinyDisplay6.cpp and TM1637TinyDisplay6.h).
But, didn't test the implications of this "fix" for negative numbers or border cases.

jasonacox, 2021-04-10:
@AndresPlazaR2 thank you!  Great catch!  The unit test for showNumber() would work because there is a polymorphic version for a num double (floating point) that would support 32 bit integers as well.   Functions showNumberDec() and showNumberBaseEx() both had explicit 16 bit integer arguments which meant that if you called them with a large value it would truncate.

I have updated showNumberDec() to handle long integers (32bit signed) and showNumberBaseEx() to handle unsigned 32bit integers.  v1.4.1 build has been created and should be picked up by Arduino in a few days.

Thanks again for submitting this issue!

jasonacox, 2021-08-14:
Closing for now as fixed in https://github.com/jasonacox/TM1637TinyDisplay/releases/tag/v1.4.1. Please re-open or comment if there are lingering issues.

closed 2021-08-14 by jasonacox