jasonacox/TM1637TinyDisplay #12: Show colon with showString

andik-ariyanto, 2021-08-21:
Can I show colon with showString function ?

jasonacox, 2021-08-21:
Currently showString renders colon as "=" on the display.  The challenge is that the library can't detect if your TM1637 display is wired to display a middle colon or decimal dots.  I've often wanted to make it work so that you could send "12:34" or "1.234" and have it translate the colon and decimal to the appropriate LED.  One option I thought about is having the class initialize with a setting for type of display (colon, decimal or none), something like this:

```c
#include <TM1637TinyDisplay.h>

#define CLK 4
#define DIO 5
#define DTYPE COLON

TM1637TinyDisplay display(CLK, DIO, DTYPE);

void setup() {
  display.setBrightness(BRIGHT_7);
}

void loop() {
  display.showString("12:34");
}
```

If DTYPE is not specified, the library would assume you want the colon rendered as text (equal sign) for scrolling etc.  If a colon is specified, it would force center and clip the values before/after the colon to fit the display (colon is in the middle), so that if you tried to send "1:2345" it would actually render " 1:23", and "1234:5" would be rendered "34:50".

Let me know if you think this would be helpful.

andik-ariyanto, 2021-08-23:
thanks for your help

jasonacox, 2021-08-24:
Hi @andik-ariyanto to be clear, I haven't update the library yet to support that.  If I did, is this the behavior you were looking for?

andik-ariyanto, 2021-08-24:
I don't think so.
What I need is like showNumberHex, but equipped with dot / colon. For example like "H."

jasonacox, 2021-08-24:
The function, `showNumberHex` has a dot parameter you can send to set the colon.   I think you could do what you want with two function calls:

```c++
#include <TM1637TinyDisplay.h>

#define CLK 4
#define DIO 5

TM1637TinyDisplay display(CLK, DIO);

void setup() {
  display.setBrightness(BRIGHT_7);

  //!        For displays with dots between each digit:
  //!        * 0.000 (0b10000000)
  //!        * 00.00 (0b01000000)
  //!        * 000.0 (0b00100000)
  //!        * 0000. (0b00010000)
  //!        * 0.0.0.0 (0b11100000)
  //!        For displays with just a colon:
  //!        * 00:00 (0b01000000)
  //!        For displays with dots and colons colon:
  //!        * 0.0:0.0 (0b11100000)

  // first set dots
  uint8_t dots = 0b01000000; // Add dots or colons - see above
  display.showNumberHex(0, dots);

  // display H at position 1
  display.showString("H", 1, 1);

}
```

jasonacox, 2021-08-24:
I have something better.  I just updated the library to v1.4.2.  It now lets you pass the dots parameter to the `showString()` function. The library will soon be available for upgrade in Arduino IDE or you can pull the latest code from this repo.  So you should be able to do this:

```c++
#include <TM1637TinyDisplay.h>

#define CLK 4
#define DIO 5

TM1637TinyDisplay display(CLK, DIO);

void setup() {
  display.setBrightness(BRIGHT_7);

  //!        For displays with dots between each digit:
  //!        * 0.000 (0b10000000)
  //!        * 00.00 (0b01000000)
  //!        * 000.0 (0b00100000)
  //!        * 0000. (0b00010000)
  //!        * 0.0.0.0 (0b11100000)
  //!        For displays with just a colon:
  //!        * 00:00 (0b01000000)
  //!        For displays with dots and colons colon:
  //!        * 0.0:0.0 (0b11100000)

  uint8_t dots = 0b01000000; // Add dots or colons - see above

  // void showString(const char s[], uint8_t length = MAXDIGITS, uint8_t pos = 0, uint8_t dots = 0);

  // display HH:SS 
  display.showString("HHSS", 4, 0, dots);
  delay(1000);
  
  // display H: 
  display.showString(" H", 4, 0, dots);
}

void loop()
{
}

```

jasonacox-sam, 2026-08-31:
This asked for colon support in `showString()`. It was implemented: library v1.4.2 added a dots parameter to `showString()` so the colon/dots can be set alongside the string.

Closing since the feature shipped. Enjoy!

— Sam 🔢

closed 2026-08-31 by jasonacox-sam