jasonacox/TM1637TinyDisplay #13: [NEWBIE] Display seconds with blink dots ?

BullzLabz, 2021-09-07:
Hello,

Thank @jasonacox for your hard work on this usefull library! 👍🏻

I would like to create a custom timer but being limited to 4 digits I would like to use the "dots" to display the seconds with blink effect/animation.

I created this animation thanks to your super tool but being beginner I can't place it in my code. Could you help me ?

`/* Animation Data - HGFEDCBA Map */
const uint8_t DOTBLINK[53][4] = {
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 0
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 1
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 2
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 3
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 4
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 5
  { 0x40, 0x00, 0x00, 0x00 },  // Frame 6
  { 0x40, 0x40, 0x00, 0x00 },  // Frame 7
  { 0x40, 0x40, 0x40, 0x00 },  // Frame 8
  { 0x40, 0x40, 0x40, 0x40 },  // Frame 9
  { 0x00, 0x40, 0x40, 0x40 },  // Frame 10
  { 0x00, 0x00, 0x40, 0x40 },  // Frame 11
  { 0x00, 0x00, 0x00, 0x40 },  // Frame 12
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 13
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 14
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 15
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 16
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 17
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 18
  { 0x40, 0x00, 0x00, 0x00 },  // Frame 19
  { 0x40, 0x40, 0x00, 0x00 },  // Frame 20
  { 0x40, 0x40, 0x40, 0x00 },  // Frame 21
  { 0x40, 0x40, 0x40, 0x40 },  // Frame 22
  { 0x00, 0x40, 0x40, 0x40 },  // Frame 23
  { 0x00, 0x00, 0x40, 0x40 },  // Frame 24
  { 0x00, 0x00, 0x00, 0x40 },  // Frame 25
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 26
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 27
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 28
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 29
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 30
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 31
  { 0x40, 0x00, 0x00, 0x00 },  // Frame 32
  { 0x40, 0x40, 0x00, 0x00 },  // Frame 33
  { 0x40, 0x40, 0x40, 0x00 },  // Frame 34
  { 0x40, 0x40, 0x40, 0x40 },  // Frame 35
  { 0x00, 0x40, 0x40, 0x40 },  // Frame 36
  { 0x00, 0x00, 0x40, 0x40 },  // Frame 37
  { 0x00, 0x00, 0x00, 0x40 },  // Frame 38
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 39
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 40
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 41
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 42
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 43
  { 0x00, 0x00, 0x00, 0x00 },  // Frame 44
  { 0x40, 0x00, 0x00, 0x00 },  // Frame 45
  { 0x40, 0x40, 0x00, 0x00 },  // Frame 46
  { 0x40, 0x40, 0x40, 0x00 },  // Frame 47
  { 0x40, 0x40, 0x40, 0x40 },  // Frame 48
  { 0x00, 0x40, 0x40, 0x40 },  // Frame 49
  { 0x00, 0x00, 0x40, 0x40 },  // Frame 50
  { 0x00, 0x00, 0x00, 0x40 },  // Frame 51
  { 0x00, 0x00, 0x00, 0x00 }   // Frame 52
};`

Would it be possible in the future to include a "blink dots" function in your Library structure?

Thank & have a good day

BullzLabz, 2021-09-08:
Hello,

I think I found a new approach to do what I want, but I'm still looking for the best way to do it.

`
void dotsBlink(uint8_t blinkDelay, uint8_t repeats, uint8_t BRIGHT_HIGH, uint8_t BRIGHT_LOW) {
  for (uint8_t i=0; i < repeats; i++) {
    setBrightness(7, false); // turn backlight off
    delay(blinkDelay);
    setBrightness(7, true); // turn backlight on
    delay(blinkDelay);
  }
  // restore backlight
  setBrightness(_backLightValue);
}
`

jasonacox, 2021-09-08:
Hi @BullzLabz, thank you for the nice words!  Are you looking to create an animation that includes using the decimal dots?  If so, yes you can.  0x80 is the value to turn on the "dot".   You can play around with the animation tool to see the different possibilities:  https://jasonacox.github.io/TM1637TinyDisplay/examples/7-segment-animator.html which I suspect you have been using. Here is 1234 with blinking dots:

```cpp
/* Animation Data - HGFEDCBA Map */
const uint8_t ANIMATION[4][4] = {
  { 0x06, 0x5b, 0x4f, 0x66 },  // Frame 0
  { 0x86, 0xdb, 0xcf, 0xe6 },  // Frame 1
  { 0x06, 0x5b, 0x4f, 0x66 },  // Frame 2
  { 0x86, 0xdb, 0xcf, 0xe6 }   // Frame 3
};
```

Based on your 2nd code example it looks like you just want to flash all the LED segments.  Your approach would also need you to set at least one segment value (brightness value is sent as part of the data update, not by itself)   I'm not where I an try this right now but I think something like this would work:

```cpp
uint8_t data[] = { 0xff, 0xff, 0xff, 0xff, 0xff, 0xff };   

void dotsBlink(uint8_t blinkDelay, uint8_t repeats, uint8_t BRIGHT_HIGH, uint8_t BRIGHT_LOW) {
  for (uint8_t i=0; i < repeats; i++) {
    setBrightness(7, false); // turn backlight off
    setSegments(data, 0, 5);
    delay(blinkDelay);
    setBrightness(7, true); // turn backlight on
    setSegments(data, 0, 5);
    delay(blinkDelay);
  }
  // restore backlight
  setBrightness(_backLightValue);
  setSegments(data, 0, 5);
}
```

Alternatively, you would keep track of the display state in a local variable (what is it displaying) so you clear and set the values on the Display, basically blinking all the dots. 

As an example, I have a [countdown timer example](https://github.com/jasonacox/TM1637TinyDisplay/blob/master/examples/TM1637-Countdown-Buttons/TM1637-Countdown-Buttons.ino) that flashes the display when countdown is reached:

```cpp
uint8_t dots = 0b01000000; // Add dots or colons (depends on display module)

if (timeElapsed >= countDown) {
  // If we hit zero - flash every second
  since = (long)((timeElapsed - countDown) / 500);
  if (since % 2) {
    display.clear();
  }
  else {
    display.showNumberDec(000000, dots, true);
  }
}
```

closed 2025-03-16 by BullzLabz