jasonacox/TM1637TinyDisplay #14: 6-Digit Animation

Noelle88, 2021-12-26:
Hi!
I think it’s a bug in the 6-digit lib.

showAnimation supports only the 4 digits in the middle, i think?

Unfortunately, my changes in the .h / .cpp from 4 to 6 did not work :-(

jasonacox, 2021-12-28:
Hi @Noelle88  you are correct, the `showAnimation()` currently only supports 4-digits.  It is on my "to-do" list to expand it to support 6-digits.  Thanks for the feedback.  It helps me add priority/motivation to make the updates. :-)

Happy new year!

Noelle88, 2022-01-01:
Okay, thanks for your feedback.
Then I'll try other functions of your lib until the showAnimation() works with 6-digits.  ;-)

I wish you a happy new year, too!

jasonacox, 2022-01-03:
Update: I have added the support for 6-digit animation, starting with the animation tool: https://jasonacox.github.io/TM1637TinyDisplay/examples/7-segment-animator6.html

The new code is on github but I need to finish some testing before I release the updated library for Arduino to pick up.  You can pull the current version from github in the meantime if you want to help test.

Noelle88, 2022-01-03:
Okay, i made the following observations during the update:

TM1637-6Digit-Test.ino  = The order of the digits is incorrect with showAnimation ().  -> 23451
All other functions will work in the right order.

I wrote my own test sketch: The decimal point should be from right to left and then a 8 from right to left.

The result: (in digit-positions from left to right)

loop 1:
Point: 4-1-5-undefined pause-3-undefined pause 
8:       1-5-undefined pause-3-1

loop 2:
Point: 3-4-1-undefined pause-5-6
8:       3-4-1-5-6-3

loop3:   different again.....

I hope I could help

jasonacox, 2022-01-03:
Thank you!  That is very helpful.  The issue was a zero index error I made in [TM1637TinyDisplay6.cpp](https://github.com/jasonacox/TM1637TinyDisplay/blob/98d06c63c47955785fd56ac2f8df7cecd925ccd0/TM1637TinyDisplay6.cpp).  I've corrected it and pushed the change.  

I also added a longer PROGMEM [animation test](https://github.com/jasonacox/TM1637TinyDisplay/blob/master/examples/TM1637-6Digit-Test/TM1637-6Digit-Test.ino) for the 6-digit display.  Not great, but least gives a better demo of the capability.

I'll run a few more tests and then release this fix as v1.4.3.

Noelle88, 2022-01-04:
YES! It works perfectly! Thank you for your great support! have a nice day!

jasonacox, 2022-01-05:
Thanks for the help!

jasonacox, 2022-05-13:
Closing this issue.  Please re-open if there is anything else we need to address.

closed 2022-05-13 by jasonacox