jasonacox/TM1637TinyDisplay #15: showNumber - call of overloaded 'showNumber(double, bool, int, int)' is ambiguous

playrobotics-alex, 2022-02-15:
Hi,

Thank you so much for this library!
I am trying to display 2 decimal numbers , one on the left side of the screen and one on the right side.
This works perfectly:
     display1.showNumber(160, false, 3, 0);    
     display1.showNumber(111, false, 3, 3);  
But this is giving me an error:
     display1.showNumber(1.60, false, 3, 0);    
     display1.showNumber(1.11, false, 3, 3);  

call of overloaded 'showNumber(double, bool, int, int)' is ambiguous

Can you please point me in the right direction?

thanks!

playrobotics-alex, 2022-02-15:
After some more testing I was able to make it partly work using:
     display1.showNumber(2.34, 2, 3, 0); 
     display1.showNumber(6.78, 2, 3, 3); 

The problem is the left part of the screen is showing 2.34 but the right part of the screen 678 (without decimal point)

jasonacox, 2022-02-16:
Hi @playrobotics-alex thanks for the finding. This could be a bug rendering the decimal for a non-zero position.  I'll take a look.

If you just send the second one, by itself, does it work?

```c
display1.showNumber(6.78, 2, 3, 3);
```
In the meantime, here a possible workaround:

```c
// assume value is < 10 and want format 0.00
double x = 2.34;
double y = 6.78;

// convert two floating point numbers to integers
int first = (int)(x*100);
int second = (int)(y*100);

// convert two numbers into a single number
long num = (first * 1000) + second;

//   void showNumberDec(long num, uint8_t dots = 0, bool leading_zero = false, 
//                 uint8_t length = MAXDIGITS, uint8_t pos = 0);
showNumberDec(num, 0b01010000, true);

```

playrobotics-alex, 2022-02-16:
Thank you for the reply, and the workaround!
The problem is I don't know both numbers at the same time, those are basically two separate timers that are stopped  independently.
The solution I ended up using was, converting the numbers to an array of chars and adding the decimal points manually:
display.showString(char_array, 3, 3, 0b10000000); 
display.showString(char_array, 3, 0, 0b10000000);

jasonacox, 2022-02-21:
Thanks for the update and I'm glad you found a workaround.  I'll see if I can fix the underlying bug as your first approach should have worked:

```cpp
display1.showNumber(2.34, 2, 3, 0);
display1.showNumber(6.78, 2, 3, 3);
```

jasonacox, 2022-02-21:
I found the bug!  There was a position offset in the decimal calculation that should not have been there.  This impacts both the TM1637TinyDisplay and TM1637TinyDisplay6 (4 and 6-digit) classes.   I will submit the fix as v1.4.4.  Thanks for the help!

```cpp
// Test run for two floating point numbers

TM1637TinyDisplay6 display(CLK, DIO);

float x = 9.99;
float y = 0.0;

void setup()
{
  display.setBrightness(BRIGHT_HIGH);
  display.clear();
}

void loop()
{
  // showNumber(num, decimal_length, length, pos)
  display.showNumber(x, 2, 3, 0);
  display.showNumber(y, 2, 3, 3);

  // move values
  x = x - 0.01;
  y = y + 0.01;
  delay(10);
}
```

playrobotics-alex, 2022-02-21:
Great job :-)
Thank you!

jasonacox, 2022-05-13:
Closing this issue.  Please re-open if there is anything else we need to address.

closed 2022-05-13 by jasonacox