jasonacox/TM1637TinyDisplay #16: How can I represent a number with a decimal point ?

grilli303, 2022-02-24:
The library is very good, but so far I have not been able to represent numbers with a decimal point. In my project, I would like to realize a temperature display and would have to display "12.3°", for example. I tried the following call: 
display.showString("\xB0", 1, 3); Test with suffix 
display.showNumber(12.3, 4, 3, 0);
However, 12:3° appears in the display. When I look at the library file, it looks like this with the command "showNumberDec" should be possible. 
Unfortunately I don't know which parameters I have to pass here. I would be very grateful for a small example. Kind regards, Daniel

jasonacox, 2022-02-25:
Hi @grilli303  - The TM1637 display modules are hard wired to display  colon ":" or  decimal "." - they cannot do both unfortunately.  You have one that is hard wired to display ":" for a clock but you need one that has decimals hardwired like this one: https://www.aliexpress.com/item/32963180147.html?spm=a2g0o.order_list.0.0.21ef1802UyE4Uc

I have several that show both, but alas, they only display colon or decimal too.  It would have been nice to have made them switchable.

jasonacox, 2022-05-13:
Closing this.  Please re-open if there is an issue we can address.

closed 2022-05-13 by jasonacox

playrobotics-alex, 2023-04-21:
@jasonacox Hi :-)
I have the same problem, 0.56 module which physically has dots and colons but probably the dots are not wired. Do you know if there is something I can do physically to the module  to wire the dots?
I found some 0.56 modules on Aliexpress which have only the dots, but I am not sure that is what I will get :-) If you have any link to such model that will help :)

jasonacox, 2023-04-23:
The problem with Aliexpress is that I sometimes order ones that advertise that they ONLY have dots and I get one that only have colons, and then vice versa. They are low cost so just keep ordering until I get what I want. :) 

As for hardware updating to fix the colon: The TM1637 drives the DP (decimal point) for each of the 7 segment displays by cycling through each of the digits. You would need to determine how the 7 segment display maps the decimal versus the colon and essentially connect the DP line from the TM1637 to the dots instead.

[image]