jasonacox/TM1637TinyDisplay #17: Fixed add negative sign for negative double

otpfiste, 2022-06-03:
Hi,

for example that is note rendered correctly:
display.showNumber(-3.1, 1, 3, 1);

The solution is to add the negative sign for negative numbers inside the for loop, see attached code which works for both the 4 and the 6 segments versions:
[showNumber-double.txt](https://github.com/jasonacox/TM1637TinyDisplay/files/8832618/showNumber-double.txt)

All the best,
Oliver

jasonacox, 2022-06-04:
Hi @otpfiste !  Great catch!  Thank you.  I can make this update but do you want to submit a pull request for this change?  It would be nice to have you listed as a contributor.

otpfiste, 2022-06-04:
Hi jasonacox,

I'm not good at git (still using svn for my projects), it's better if you make the update, I will then test it if you want.

Thanks,
Oliver

jasonacox, 2022-06-04:
No problem!  Here is the diff (from vscode).  I'll update and test for both 4 and 6 digit functions.

[image]

Thanks again!

otpfiste, 2022-06-04:
Yes, it's correct.

Thanks again for your nice library,
Oliver

jasonacox, 2022-06-04:
Thanks, Oliver!  I am running test on 4 and 6 digit displays now.  I have uploaded the revision to github.  I noticed that there was a condition for negative non-floating point numbers that was also not handled so I fixed that as well.

If you can, please test (see https://github.com/jasonacox/TM1637TinyDisplay/commit/44ac84c700fa7ffae93631a7d1357de973f90f83).  I will upload to the patched version as v1.5.1 for Arduino later today.

otpfiste, 2022-06-05:
I updated to v1.5.2 in Arduino IDE and tested it with my 4 and 6 segs displays, all works great!

Note: I re-checked the source code and found several comments to update in TM1637TinyDisplay6.h file: 
(0 - leftmost, 3 - rightmost)  -->   (0 - leftmost, 5 - rightmost)

All the best,
Oliver

jasonacox, 2022-06-05:
Thanks, Oliver!  I updated the comments as you mentioned.

jasonacox-sam, 2026-08-31:
This reported negative decimals (e.g., `showNumber(-3.1, 1, 3, 1)`) not rendering the minus sign, with a fix contributed by the reporter. It was incorporated into v1.5.2 (also fixing negative non-floating point numbers), tested by the reporter on both 4- and 6-digit displays, and the doc comments were cleaned up too.

Closing since the fix shipped in v1.5.2. Thanks for the contribution!

— Sam 🔢

closed 2026-08-31 by jasonacox-sam