jasonacox/TM1637TinyDisplay #19: setBrightness : enhancement

stef-ladefense, 2022-07-31:
I noticed that in the code, when you change the brightness or simply turn off the display, you have to do a setBrightness followed by rewriting the message.
It's counterproductive.
If for example I want to flash the display, at worst I have to do either:
```
for (uint8_t i = 0; i < 5; i++) {
display.setBrightness(7, false);
display.showNumber(_num, true);
delay(250);
display.setBrightness(7, true);
display.showNumber(_num, true);
delay(250);
}
```
or better not using setBrightness but rewriting the message all the same
```
for (uint8_t i = 0; i < 5; i++) {
display.clear();
delay(250);
display.showNumber(_num, true);
delay(250);
}
```
By simply moving the writing of the TM1637_I2C_COMM3 into the setBrightness, the command becomes instantaneous and allows the brightness to be varied or the display to be turned off or on without rewriting a message after the setBrightness.
this command doesn't need anything else to work.
```
display.showNumber(_num, true);
for (uint8_t i = 0; i < 5; i++) {
display.setBrightness(7, false);
delay(250) ;
display.setBrightness(7, true);
delay(250) ;
}
```
change in TM1637TinyDisplay.cpp
```
void TM1637TinyDisplay::setBrightness(uint8_t brightness, bool on)
{
  m_brightness = (brightness & 0x07) | (on? 0x08 : 0x00);
  
  start();
  writeByte(TM1637_I2C_COMM3 + (m_brightness & 0x0f));
  stop();
}

void TM1637TinyDisplay::setSegments(const uint8_t segments[], uint8_t length, uint8_t pos)
{
...
  // Write COMM3 + brightness
  //start();
  //writeByte(TM1637_I2C_COMM3 + (m_brightness & 0x0f));
  //stop();
}
```
tested on my current project and works without any observed side effects.
What do you think ?

Stephane

jasonacox, 2022-07-31:
I love it!  ❤️  You are absolutely right.  This is more intuitive.  

Do you want to submit a pull request with your changes?  I can make the change myself but would prefer if you get the credit.

stef-ladefense, 2022-07-31:
thank ❤
voila !

jasonacox, 2022-08-01:
Thank you!  This is a great upgrade! This will be part of release v1.6.0, coming soon.

jasonacox, 2022-08-02:
v1.6.0 Released: https://www.arduino.cc/reference/en/libraries/tm1637tinydisplay/

You code is now live.  Thanks for your contribution and help!

jasonacox-sam, 2026-08-31:
This requested making `setBrightness()` preserve the displayed content so you can flash/dim the display without rewriting it. Implemented and released in v1.6.0, with credit to the contributor.

Closing since the enhancement shipped.

— Sam 🔢

closed 2026-08-31 by jasonacox-sam