jasonacox/TM1637TinyDisplay #22: ENHANCEMENT: Non-blocking animations/scrolling

hackerceo, 2022-08-30:
Hi,  I am looking for this functionality in a non-blocking design.  Is there any plans to do this? Is there interest if I do this and issue a pull request?

Thanks,
Nick

jasonacox, 2022-08-31:
Hi @hackerceo - Thanks for opening this issue. Great question and yes, I would be happy to accept a PR that gives us that enhancement! 🙏 

@Kravatox I'll move your question to a new issue as it is not related to animation/scrolling: See #23

Frtrillo, 2022-09-16:
Hello, at least non-blocking scrolling text would be very useful IMO

hackerceo, 2022-09-16:
I plan to work on this Sept-Oct.  It should play animations AND scroll text in a non-blocking way.

jasonacox, 2022-09-18:
Thanks for the PR here https://github.com/jasonacox/TM1637TinyDisplay/pull/24 !  I can't wait to use this for my projects.

I have one minor change suggestion for API consistency: 

Can we change the parameters for `scrollString()` and `startAnimation()` where we put content payload first and put `bool usePROGMEM = false` at the end with a default to false?  This would align with the rest of the function.  We could even add `scrollString_P()` and `startAnimation_P()` as alias functions that call their main functions with usePROGMEM = true.  

Seems like that would help with consistency, but I could be convinced otherwise.  What do you think?

```c++
  void startAnimation(const uint8_t (*data)[4], unsigned int frames = 0, unsigned int ms = 10, bool usePROGMEM = false );
  void scrollString(const char s[], unsigned int ms = DEFAULT_SCROLL_DELAY, bool usePROGMEM = false);
```

hackerceo, 2022-09-18:
One last thing. What do you think about renaming "scrollString" function to
"startStringScroll"? I think having the functions which start non-blocking
operations begin their name with "start" makes it a little more clear that
those functions only start the process. One could assume "scrollString"
will handle everything for you.

jasonacox, 2022-09-18:
I like it!  Great thinking @hackerceo ...  And it is consistent with `startAnimation()`. Do you want to make these changes or should I?  I can merge and update or can wait to merge until after you commit the changes to the same PR.

hackerceo, 2022-09-18:
I'll make the changes in a few hours.

hackerceo, 2022-09-19:
@Frtrillo you should be happy as non-blocking animation/scrolling functionality is now added to this library!

jasonacox, 2022-09-19:
I'll release this as v1.7.0 for the [Arduino library](https://www.arduino.cc/reference/en/libraries/tm1637tinydisplay/) and add a TODO to port this to the 6-digit display as well.

Thanks for this great enhancement @hackerceo !

hackerceo, 2022-09-19:
Sounds great! I think if we can mention non-blocking animations that would
attract adoption and use of the library.  I think this is the only TM1637
library that does that.

jasonacox, 2022-09-20:
Agree. You added that to `library.properties` so it now shows up on the [Arduino library](https://www.arduino.cc/reference/en/libraries/tm1637tinydisplay/) page:

[image]

The update announcement was broadcast on twitter -- I added comments: https://twitter.com/jasonacox/status/1572097524223311872?s=20&t=KQt712j4CPpizeAMyt8vng

Thanks again @hackerceo !

hackerceo, 2022-10-11:
I agree with your suggestion about the usePROGMEM parameter. I could even
agree with adding a second set of functions ending in "_P" but I think it
would be cleaner to have a single set of functions that do something rather
than 2 sets of functions that do the same thing but are passed data that
needs to be processed in a slightly different way.

Do you want to make the changes or should I?

I also noticed that you are using a build process for checking for breaks.
The new non-blocking methods are not in the test.ino sketch yet.

Best Regards,
Nick Benik

hackerceo, 2023-04-01:
I have submitted https://github.com/jasonacox/TM1637TinyDisplay/pull/31 which adds support for non-blocking animations on 6 digit displays.  Once PR #31 is merged in you can close this issue.