jasonacox/TM1637TinyDisplay #23: Using 4 and 6 digit displays at the same time

jasonacox, 2022-08-31:
Thanks for the answer, I'm a beginner with the Arduino code and I will not be able to help you, sorry.
I have another question, in my code I use two display, one with 4 digits and another with 6 digits, and when I include TM1637TinyDisplay.h library and TM1637TinyDisplay6.h I have this error : "warning: "MAXDIGITS" redefined" and "warning: "FRAMES" redefined". Is there any solution to use both library the same time ?

CaptainSMan, 2023-11-10:
Is it possible to share the one CLK pin for say two 6digit displays?

jasonacox, 2023-11-13:
That is possible. However, as you update one display, it is unknown what will happen to the the other display. It is possible that it will zero out even with no DIO toggles. Let us know if you try it out.

Alternatively, you could create a new class that manages two displays and update both displays at the same time using a shared CLK.
* Update [writeByte()](https://github.com/jasonacox/TM1637TinyDisplay/blob/f13913e803ccfbe56f9ac924371738d19d66a597/TM1637TinyDisplay.cpp#L741) to receive both display A and B byte and write at same time
* Update [writeBuffer()](https://github.com/jasonacox/TM1637TinyDisplay/blob/f13913e803ccfbe56f9ac924371738d19d66a597/TM1637TinyDisplay.cpp#L216) to send both display buffers to writeByte()

I don't think it would be too difficult. If you or anyone else wants to attempt adding this new class, I would be happy to accept a PR to add it to the library.

CaptainSMan, 2023-11-13:
Thanks Jason.
I have kept them separate for now in the interest of getting my project up and running, might revisit this in the next iteration.

jasonacox-sam, 2026-08-31:
This thread covered using the 4-digit and 6-digit libraries together (the `MAXDIGITS`/`FRAMES` redefinition warnings) and sharing a single CLK pin across displays. Guidance given: keep separate instances/pins where possible, or build a manager class driving both — sharing one CLK risks clobbering the idle display's state.

Closing since the questions were answered. Re-open if anyone builds a dual-display class worth sharing.

— Sam 🔢

closed 2026-08-31 by jasonacox-sam