jasonacox/TM1637TinyDisplay #26: Error compile with 1.7 and esp8266 core 3.02

stef-ladefense, 2022-10-29:
Hi,

have error ...
idea ?

D:\Arduino\libraries\TM1637TinyDisplay\TM1637TinyDisplay.cpp: In member function 'void TM1637TinyDisplay::startAnimation(const uint8_t (*)[4], unsigned int, unsigned int, bool)':
D:\Arduino\libraries\TM1637TinyDisplay\TM1637TinyDisplay.cpp:641:28: warning: invalid conversion from 'const uint8_t (*)[4] {aka const unsigned char (*)[4]}' to 'uint8_t (*)[4] {aka unsigned char (*)[4]}' [-fpermissive]
     m_animation_sequence = data;
                            ^~~~
D:\Arduino\libraries\TM1637TinyDisplay\TM1637TinyDisplay.cpp: At global scope:
D:\Arduino\libraries\TM1637TinyDisplay\TM1637TinyDisplay.cpp:651:100: warning: default argument given for parameter 3 of 'void TM1637TinyDisplay::startStringScroll(const char*, unsigned int, bool)' [-fpermissive]
 void TM1637TinyDisplay::startStringScroll(const char (*s), unsigned int ms, bool usePROGMEM = false) {
                                                                                                    ^
In file included from D:\Arduino\libraries\TM1637TinyDisplay\TM1637TinyDisplay.cpp:29:0:
D:\Arduino\libraries\TM1637TinyDisplay/TM1637TinyDisplay.h:434:8: note: previous specification in 'void TM1637TinyDisplay::startStringScroll(const char*, unsigned int, bool)' here
   void startStringScroll(const char s[], unsigned int ms = DEFAULT_SCROLL_DELAY, bool usePROGMEM = false);
        ^~~~~~~~~~~~~~~~~
D:\Arduino\libraries\TM1637TinyDisplay\TM1637TinyDisplay.cpp: In member function 'void TM1637TinyDisplay::startStringScroll(const char*, unsigned int, bool)':
D:\Arduino\libraries\TM1637TinyDisplay\TM1637TinyDisplay.cpp:677:26: warning: invalid conversion from 'const char*' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
     m_animation_string = s;
                          ^

jasonacox, 2022-11-04:
Thanks for opening this @stef-ladefense !   This was not showing up compiling for Uno but seems to be across all ESP8266 cores.  I've added the required type casting and removed erroneous defaults in the functions (should have only been in the TM1637TinyDisplay.h definitions).  It now compiles.

@hackerceo Can you also verify these changes work with your use case?  Commit =  https://github.com/jasonacox/TM1637TinyDisplay/commit/8720b959ba89ffac256ed52fd6ee3aa9f70aa9e3

stef-ladefense, 2022-11-04:
hi @jasonacox 

it's ok for my code, compiling ok with esp8266 core 3.02

very good !

hackerceo, 2022-11-04:
@jasonacox The changes look fine.  I just have one question on the declaration of TM1637TinyDisplay::startStringScroll, there is no default value set any more for the usePROGMEM parameter.  Does that mean that calling startStringScroll without setting that parameter will have it default to false?

jasonacox, 2022-11-05:
Thanks @stef-ladefense !  That's good news.

And thanks, @hackerceo !  - All the defaults are actually set in the header file, `TM1637TinyDisplay.h` instead of the .cpp file which is what was causing the compiler warning for ESP8266.  I don't know why it would be treated different for an Arduino AVR compile. Anyway, to your specific question on `startStringScroll()`, the header file has the `usePROGMEM` default set to false which is what is driving he behavior:

```cpp
  //! The function used to begin a non-blocking scroll of a string
  //!
  //! @param usePROGMEN Indicates if the passed string data is coming from a PROGMEM defined variable
  //! @param ms Time to delay between each frame
  void startStringScroll(const char s[], unsigned int ms = DEFAULT_SCROLL_DELAY, bool usePROGMEM = false);
  void startStringScroll_P(const char s[], unsigned int ms = DEFAULT_SCROLL_DELAY);
```

stef-ladefense, 2022-11-05:
> Merci d'avoir ouvert ce@stef-ladefense! Cela n'apparaissait pas lors de la compilation pour Uno, mais semble concerner tous les cœurs ESP8266. J'ai ajouté le casting de type requis et supprimé les valeurs par défaut erronées dans les fonctions (n'aurait dû figurer que dans les définitions TM1637TinyDisplay.h). Il compile maintenant.
> 
> @hackerceoPouvez-vous également vérifier que ces modifications fonctionnent avec votre cas d'utilisation ? Commettre = [8720b95](https://github.com/jasonacox/TM1637TinyDisplay/commit/8720b959ba89ffac256ed52fd6ee3aa9f70aa9e3)

I didn't dig anymore but one weird thing with 1.7 version.
in the Arduino\libraries directory, I deleted the folder of your library, then put back with the rar 1.70.

compile with arduino ide
under AVR uno, no problem.
change to esp8266 (lolin D1) error
reset AVR uno: same error

I didn't understand why, even clearing the cache.

but hey, the error is corrected, that's the main thing

jasonacox, 2022-11-06:
That's interesting!  I did some more digging.  The fundamental difference between AVR and ESP8266 builds is that the IDE uses a different compiler in the tool chain. The ESP8266 uses a Tensilica L106 32-bit RISC processor and the Uno uses a Microchip ATmega328P microcontroller with these compilers:

* AVR = avr-g++
* ESP8266 = xtensa-lx106-elf-gcc

This makes sense now. However, your example doesn't make sense. I couldn't replicate that.  As you say, main thing is that error is corrected but is very puzzling.

stef-ladefense, 2022-11-06:
i'im ok too ! puzzling !
i close this issue

closed 2022-11-06 by stef-ladefense

jasonacox, 2022-11-06:
Thanks again for finding and raising this issue @stef-ladefense 🙏