jasonacox/TM1637TinyDisplay #34: Platform.io. shows only version 1.1.0 on M1 processors

Mir1001, 2023-07-01:
Searching for the library in Platform.io (Mac with M1 processor, VC) shows only the 1.1.0 version (the changelog in VC shows all versions). If I try to update it manually (platformio.ini), I get a following error:

```
Library Manager: Removing TM1637TinyDisplay @ 1.1.0
Library Manager: TM1637TinyDisplay@1.1.0 has been removed!
Library Manager: Installing jasonacox/TM1637TinyDisplay @ ^1.10.0
Error: Could not find the package with 'jasonacox/TM1637TinyDisplay @ ^1.10.0' requirements for your system 'darwin_arm64'
```

It looks/guess same problem as [issue in similar lib.](https://github.com/earlephilhower/arduino-pico/issues/661)

Mir1001, 2023-07-01:
**Workaround to use latest version of the lib** 
Import lib with git repro in platformio.ini:
```
lib_deps = 
	...
	https://github.com/jasonacox/TM1637TinyDisplay.git
        ...
```

jasonacox, 2023-07-01:
Hi @Mir1001 !  

A quick search and I find a similar (and old) thread for another package: https://community.platformio.org/t/apple-silicon-m1-chip-darwin-arm64-problem-with-xtensa32-toolchain/19270/3  - It seems the solution was to have a PlatformIO team member update the pacakge. I'm not familiar enough with the PlatformIO stack to know how to request them to compile the newer version.  

I see our library in the registry here: https://registry.platformio.org/libraries/jasonacox/TM1637TinyDisplay - I'll hunt around to see how we can request that they update to latest.

Thanks for opening this!

jasonacox, 2023-07-01:
I found https://docs.platformio.org/en/latest/librarymanager/creating.html which indicates that a version element in the manifest. I added that in hopes that it will cause the PlatformIO scanner to pick up and build the latest.

jimb-cc, 2024-03-18:
I'm seeing this problem too in windows11 but the workaround to pull in the .git library is working fine for me.

jasonacox-sam, 2026-09-06:
This reported PlatformIO only offering v1.1.0 of the library (M1 Macs, also seen on Windows 11) — a PlatformIO registry indexing issue rather than a library bug. The workaround (confirmed working by two people in this thread) is to depend on the git repo directly in `platformio.ini`:

```
lib_deps = https://github.com/jasonacox/TM1637TinyDisplay.git
```

Jason also added an explicit version element to the library manifest to help the registry scanner. Closing due to inactivity — the git workaround is solid, and the library itself is now up to [v1.12.2](https://github.com/jasonacox/TM1637TinyDisplay/releases/tag/v1.12.2).

— Sam 🔢

closed 2026-09-06 by jasonacox-sam