jasonacox/TM1637TinyDisplay #36: Will not accept an unsigned int

pxgator, 2023-12-16:
Hi, I am using your lib on a 6 digit display. I am using #include <TM1637TinyDisplay6.h>
When I assign an int value to a variable such as
int zx = 9999;
display.showNumber(zx); 

This works fine but when I do

unsigned int zx = 999999;
display.showNumber(zx); 

I get the compile error:

Compilation error: call of overloaded 'showNumber(unsigned int&)' is ambiguous

Any workarounds?

pxgator, 2023-12-16:
P.S. 
I wrote PICAXE code for these displays and discovered that the sequencing command
of 0xC3 allows logical sequencing but it is from right to left. As you already know the
command 0xC0, which allows left to right sequencing on the 4 digit displays, does not
work with the 6 digit displays.

jasonacox, 2023-12-16:
Hi @pxgator  - I can't replicate this. What board are you using?

I suspect that the platform you are using has `unsigned int` as a 16 bit value which can only hold values up to 65,535 .  To hold 999,999 you will need to use a `long int`:

```c
// Includes
#include <Arduino.h>
#include <TM1637TinyDisplay6.h>

// Module connection pins (Digital Pins)
#define CLK 4
#define DIO 5

// Initialize TM1637TinyDisplay - 4 Digit Display
TM1637TinyDisplay6 display(CLK, DIO);

void setup() {
  // put your setup code here, to run once:
  display.begin();
}

void loop() {
  // put your main code here, to run repeatedly:
  long zx = 999999;
  display.showNumber(zx);
}
```

Or it could be that your platform doesn't see a difference between `int` and `long` so it can't match these overloaded functions:

https://github.com/jasonacox/TM1637TinyDisplay/blob/f13913e803ccfbe56f9ac924371738d19d66a597/TM1637TinyDisplay6.h#L220-L247

pxgator, 2023-12-16:
Thank you Sir, your solution works fine. Thank you for your time and great library.

Best Regards

jasonacox, 2023-12-16:
Awesome! Good job. I'll close this for now. Feel free to respond or open a new issue if needed.

closed 2023-12-16 by jasonacox