jasonacox/TM1637TinyDisplay #39: Enhanced character set

J-Brinkman, 2024-02-16:
[image]

Dear Jason,
I love your tiny display library, it is excellent! However I think the character set could be improved and suggest the following changes:
Sym	Hex	Proposal		Current
a	5F	01011111	01110111
e	7B	01111011	01111001
i	10	00010000	00000100
j	0E	00001110	00011110
k	70	01110000	01110110
K	72	01110010	01110110
{	46	01000110	00111001
l	30	00110000	00110000
}	70	01110000	00001111
~	04	00000100	01000000
≡	49	01001001	addition
°	63	01100011	addition
¬	44	01000100	addition
º	6B	01101011	addition
ǁ	36	00110110	addition
@	7B	01111011	addition
˽	1C	00011100	addition
’	02	00000010	addition
<	61	01100001	00111001
>	43	01000011	00001111
≤	69	01101001	addition
≥	4B	01001011	addition
M	33	00110011	-
m	27	00100111	-
Mm	Upper M		See attached file. Makes a proper upper M
nn	Lower m		See attached file. Makes a proper lower m
W	1E	00011110	-
w	3C	00111100	-
Ww	Upper W		See attached file. Makes a proper upper W
uu	Lower w		See attached file. Makes a proper lower w

Would you please consider changing this? Would be very, very nice!

Jeroen Brinkman

jasonacox, 2024-02-17:
HI @J-Brinkman !  I'm open to this. Do you want to submit a PR?

J-Brinkman, 2024-02-18:
Dear Jason,

I have never ever submitted a PR, so have mercy on me. I just tried to read the documentation on pull requests, but it was far from trivial. Trying to submit a PR, doesn’t give me any clues too. I just couldn’t find what to do. 
I did find the ASCI encoding(s) in the library and concluded adding an extended ASCI set has limited value. So I only changed the regular set. I enclosed this changed set in the attached file. Is this workable for you? Otherwise, what do you want me to do?

[ASCI table.txt](https://github.com/jasonacox/TM1637TinyDisplay/files/14318994/ASCI.table.txt)
[Example showString.txt](https://github.com/jasonacox/TM1637TinyDisplay/files/14319273/Example.showString.txt)

Jeroen Brinkman

jasonacox, 2024-04-25:
Hi @J-Brinkman !  I'm sorry for the delay. I had quite a bit of travel and this got buried in my backlog.

I will create a branch with your updated ASCI table for review. Thanks for doing this!

jasonacox, 2024-07-15:
Implemented in v1.11.0

closed 2024-07-15 by jasonacox