jasonacox/TM1637TinyDisplay #41: Flip Display for Temperature

jasonacox, 2024-09-27:
Incoming question:

> I saw your library TM1637TinyDisplay for 7 segment display and I need help if you have some free time.
> I want to display temperature, and I want to turn the display upside down so I can use dot as degree and fourth digit to display C.
> So I want to display 'C on  the fourth segment all the time.
> Can you help me to do that?

For degree C, I usually use the built-in degree mark - See code example: https://github.com/jasonacox/TM1637TinyDisplay/blob/a4d07fba0cef792d5c87a0cac013a9f28070146f/examples/TM1637Demo/TM1637Demo.ino#L384-L389

[image]

If you want to flip the display, use the function: flipDisplay(..) 
See: https://github.com/jasonacox/TM1637TinyDisplay/blob/a4d07fba0cef792d5c87a0cac013a9f28070146f/TM1637TinyDisplay.h#L136-L142

To set the “dot” you can use showNumberDec(..)
See: https://github.com/jasonacox/TM1637TinyDisplay/blob/a4d07fba0cef792d5c87a0cac013a9f28070146f/examples/TM1637Test/TM1637Test.ino#L220-L224

jasonacox, 2024-10-11:
> I think you did not realize what I really wanted. I want to use the decimal dot as a degree symbol. I did it with 2 digits, but 

> when i use 3 digits dot disappears. 
> You can see it in the images below and you can see code too. 
> 
> [image]
> [image]
> 
> ```python
> #include <Arduino.h>
> #include <TM1637TinyDisplay.h>
> 
> // Define the connections pins
> #define CLK 9
> #define DIO 8
> 
> 
> TM1637TinyDisplay display(CLK, DIO);
> 
> void setup() {
>    display.setBrightness(4); // Set brightness level (0-7)
>   
>   // Flip the display upside down
>   display.flipDisplay(true);
>   
>   // Display 'C' with a decimal dot on the 4th digit
>   uint8_t segments[] = {0x00, 0x00, 0b10000000, 0b00111001 }; 
>   display.setSegments(segments);
> }
> 
> void loop() {
> 
>   display.showNumber(22, false, 2, 0);  
>   delay(1000); // Wait for 1 second
>   
> 
>   display.showNumber(111, false, 3, 0);  
>   delay(1000); // Wait for 1 second
> 
> }
> ```
> 
>

jasonacox, 2024-10-11:
Hi Filip,

You actually need to set the decimal on the third number, not on the "C". When you do the second showNumber() call, it erases the decimal. 

You need to set the decimal when you set the number:

```cpp
  k = 3; // Location for decimal
  display.showNumberDec(111, (0x80 >> k), false, 3, 0);
```

jasonacox-sam, 2026-09-06:
This was a how-to question about displaying temperature upside-down with the decimal point doubling as a degree symbol. Jason's answer: set the decimal point on the *third* digit at the time you write the number, since a second `showNumber()` call erases it:

```cpp
k = 3; // Location for decimal
display.showNumberDec(111, (0x80 >> k), false, 3, 0);
```

Closing as answered. More display examples live in the repo: https://github.com/jasonacox/TM1637TinyDisplay/tree/main/examples

— Sam 🔢

closed 2026-09-06 by jasonacox-sam